Duet3D/PluginRepository #329: Submission error in process

MintyTrebor, 2022-08-18:
When adding a new submission you receive 2 msgs/emails.
The first reports an error : "Run Failed: remove-plugin - master (7f4b230)"
The second is a breakdown of the plugin submission with each step's completion status.

I am assuming the first msg is in error when submitting a new plugin, as there is nothing to remove.

MintyTrebor, 2022-08-18:
I now have received Run Failed emails - even though the submission was successful

yasasw, 2022-08-18:
Hi @MintyTrebor 
Congratulations on submitting your plugins. They are now listed in https://plugins.duet3d.com/

I see. Let me take a look. Can you tell me the issue number?

MintyTrebor, 2022-08-18:
I received a failure msg for both submissions:
[Duet3D/PluginRepository] Run failed: submit-new-plugin - master (0d1f4dc)
[image]
[Duet3D/PluginRepository] Run failed: remove-plugin - master (0d1f4dc)
[image]

MintyTrebor, 2022-08-18:
Issues 325 & 327

MintyTrebor, 2022-08-18:
Also if the README.md file contains images from the plugins repo, they do not display:
[image]

I think this is because its using a url path that is only reachable when using the github website. I have updated BtnCmd README.md to use the following format:

https://raw.githubusercontent.com/<username>/<repo name>/<branch>/<path>/<img name>
eg
https://raw.githubusercontent.com/MintyTrebor/BtnCmd/main/wikires/v0.8.13_BtnCmd_MainWindow_1.png

If its possible to re-pull the README.md to update https://plugins.duet3d.com/plugins/BtnCmd.html we can test if the changes I made works. If it does - it might be worth while adding a section to the submission guide, which describes how to avoid the problem.

yasasw, 2022-08-18:
Hi @MintyTrebor 
The errors were thrown as the current workflows(both submission/removal) are being triggered for opened issues (and workflow for removal will be force-stopped if the issue was opened for a plugin submission).

yasasw, 2022-08-18:
Hi @MintyTrebor 
We are still working on the content-updating workflows. Atm, need to remove and resubmit the plugin.
Also, thanks for raising this. I will update the documentation mentioning that the links in README.md should be updated.

I have removed the plugin.
Can you lodge a new submission for BtnCmd?

yasasw, 2022-08-18:
Thanks and cheers!

closed 2022-08-18 by yasasw

MintyTrebor, 2022-08-18:
Confirmed! the URL structure I posted above does work, and images display properly

yasasw, 2022-08-18:
Thanks for confirming @MintyTrebor